board707/w80x_arduino #1: PWM error

AnatolSher, 2023-02-02:
Hello! PWM work not correctly. 
`// 1 - For Arduino PWM need set PWM_OUT_INVERSE_DISABLE 
// 2 - MultiPWM not worked. 

#include <Arduino.h>
#include <stdio.h>

#define DUTY_MAX 250
#define DUTY_MIN 50

int pwm_pins[3] = { PB0, PB1, PB2 };

void setup() {
  //put your setup code here, to run once:
  pinMode(pwm_pins[0], PWM_OUT);
  pinMode(pwm_pins[1], PWM_OUT);
  pinMode(pwm_pins[2], PWM_OUT);
  
  analogWrite(pwm_pins[0], DUTY_MAX);  
  analogWrite(pwm_pins[1], DUTY_MAX);
  analogWrite(pwm_pins[2], DUTY_MAX);
}

void loop() {
  // put your main code here, to run repeatedly:

  for (int pw = 0; pw < 3; pw++) {

    for (int i = DUTY_MAX; i > DUTY_MIN; i--) {
      analogWrite(pwm_pins[pw], i);
        delay(1);
    }
    for (int i = DUTY_MIN; i < DUTY_MAX; i++) {
      analogWrite(pwm_pins[pw], i);
      delay(1);
    }
  }
}
`

board707, 2023-02-03:
On what board did you test it - W806 or W801 ?

AnatolSher, 2023-02-03:
W806-Kit v1.0

board707, 2023-02-03:
Perhaps there was some misunderstanding between us :)
I have nothing to do with the manufacturers of these boards, for me it's just a hobby. So my knowledge is very limited. I only use what I found in the Net.

At the time I don't have an 806 board, only an 801. I will try to figure out the SDK for 806, but if you find something yourself, please write :)

AnatolSher, 2023-02-03:
I understood you. It's also a hobby for me. I have an 801 board. And I will check all PWM channels on it.

AnatolSher, 2023-02-03:
I checked. The W801 board has the same behavior. Maybe the pin multiplexing is wrong.
А по русски здесь писать можно?

board707, 2023-02-04:
Можно :)
Но полезнее все же по английски, так как больше народу смогут участвовать.

board707, 2023-02-04:
Do you test for the same issue under windows?

AnatolSher, 2023-02-05:
This is independent of the operating system. I have tested compiling and uploading a PWM sketch on both Windows and Linux. Arduino IDE 1.8.19

AnatolSher, 2023-02-07:
Странные китайские программисты... :) 
Странная концепция инициализации пинов в различных режимах... :)
Чтобы победить ошибку пришлось сделать так - при каждом вызове analogWrite() инициализировать PWM
Лишнее за комментировал.

Strange Chinese programmers... :)
Strange concept of initializing pins in different modes... :)
To defeat the error, I had to do this - with each call to analogWrite () initialize PWM.
Banned unnecessary program lines of the function.

[image]

AnatolSher, 2023-02-20:
Привет! Разобрался я почему PWM в многоканальном и одновременном исполнении не работал. Пришлось всю инициализацию переписать. Чуть позже выложу исходники

board707, 2023-02-23:
Привет
Я до конца недели в отьезде

AnatolSher, 2023-03-19:
Исправлено в HAL 0.6.0

closed 2023-03-19 by AnatolSher