board707/w80x_arduino #7: HAL PWM example

AnatolSher, 2023-02-05:
For HAL ver "0.4.0" Tested. Add to example folder pls
[hal_pwm_led.ZIP](https://github.com/board707/w80x_arduino/files/10611531/hal_pwm_led.ZIP)

board707, 2023-02-06:
Thank you.
A little note on the code
Since the duty formula is like this:

`Duty Cycle = (19 + 1) / (99 + 1) = 20% ` 

the value of DUTY_MAX should be 99, not 100.

Is anywhere a documentation for the HAL ver "0.4.0" ?

AnatolSher, 2023-02-06:
Я не знаю какое значение должен иметь DUTY_MAX. Примеры по тестированию периферии я не придумываю сам. Это адаптация кода из папки DEMO стандартного SDK. Но насколько я понял, при беглом прочтении, из даташита (reference manual) на 80х серию, PWM там 8 битный. Т.е коэффициент заполнения может принимать значения от 0 до 0xFF. Эта формула имеет отношение к инициализации PWM модуля. Там даже есть немного пояснений в виде
PWM Frequency = 40MHz / Prescaler / (Period + 1)；
Duty Cycle(Edge Aligned)   = (Pulse + 1) / (Period + 1)
Duty Cycle(Center Aligned) = (2 * Pulse + 1) / (2 * (Period + 1))
Но я глубоко не вникал в эти подробности. Единственно, на что обратил внимание, что HAL "0.6.0" имеет более качественный и понятный код нежели "0,4,0". 
Никакой документации на HAL , как например STM делает, я не нашел. Препарирую внутренности при помощи CDK и CKLink-Lite.

board707, 2023-02-06:
Could you please show the link to SDK ?

AnatolSher, 2023-02-06:
https://cloud.mail.ru/public/D7Be/aG96BRcwZ
https://github.com/henrycoding/cklink-lite

AnatolSher, 2023-02-06:
Сейчас проверил осциллографом выход PWM канала. От 0 до FF картинка заполнения на экране четкая. PWM 8 битный. 

Now I checked the output of the PWM channel with an oscilloscope. From 0 to FF, the filling pattern on the screen is clear. PWM is 8 bit.

board707, 2023-02-06:
> Сейчас проверил осциллографом выход PWM канала. От 0 до FF картинка заполнения на экране четкая. PWM 8 битный.

Это верно, но не имеет отношения к вашему примеру hal_pwm_led
В примере вы устанавливаете сброс таймера по совпадению на значении 99, а значит и максимальное значение регистра сравнения (duty) уже не может быть более 99.
Мое замечание было только об этом, я не собирался подвергать сомнению пример в целом.

Я, кстати. запустил его на реальной плате и посмотрел сигналы логическим анализатором... это то как я увидал эту ошибку.

board707, 2023-02-06:
Thank very much for the links :)

AnatolSher, 2023-02-06:
Возможно я еще что то не догоняю... Но установка коэффициента заполнения приводи в конечном итоге к вызову функции
__PWM_Duty_Config с прямой установкой значения в регистр CMPDAT. Поэтому, мне кажется, не важно что там китайцы записали в hpwm->Init.Pulse в процессе инициализации. 
[image]

board707, 2023-02-06:
> не важно что там китайцы записали в hpwm->Init.Pulse в процессе инициализации.

 Тем не менее, Вы можете посмотреть сами - установка заполнения 99 в вашем примере дает постоянный высокий сигнал на выходе, то есть duty = 100%

Вот эта формула довольно однозначна:
`Duty Cycle(Edge Aligned) = (Pulse + 1) / (Period + 1)`
Чтобы получить на выходе Duty Cycle = 100, надо подставить в нее 99.

AnatolSher, 2023-02-06:
Ок. Возможно... Что то мне подсказывает что надо еще и периодом манипулировать чтобы получить нужную дискретность заполнения в диапазоне от 0 до FF. Требует глубокого осмысления китайских иероглифов :)

board707, 2023-02-06:
Закрываю.

closed 2023-02-06 by board707