board707/w80x_arduino #10: w801 pinout.

AnatolSher, 2023-02-07:
[image]

AnatolSher, 2023-02-07:
Это nictrace c изиэлектроникс сделал. Я у него дернул. Копирайт не нарушаем :)
Дополнительные ряды - это для отладчика. Кто доползет до такого уровня - разберется.
Вот тут его статья. Он там про пины пишет. Я бегло сравнивал даташиты. Есть разница для ремапинга.

board707, 2023-02-07:
Спасибо огромное, я этой статьи еще не видел. Сразу цитата из нее:

>код, скомпилированный под W801 не работает на W806. Дело в разных адресах старта образа. 

А у нас вроде работает? При настройке прошивки под линукс я брал адреса образа из СДК для 806- ого - и он отлично работает на 801м. Вы проверяли - после моих изменений прошивка 806-ого не перестала работать?

AnatolSher, 2023-02-07:
Ну, все правильно... Мы то пользуемся сборкой SDK 806 под ардуино. C 806 -> 801 запускается. Если в CDK  я наваяю проект для 801 с его SDK, прошивка на 806 не взлетит.

board707, 2023-02-07:
Еще одно - я в настройках в момент прошивки добавил сброс по RTS.  Для 801-ого это очень удобно. При прошивке 806 это не мешает?  Можно сделать отдельные настройки на каждый чип.
Кстати, и адреса загрузки можно тоже настроить разные - это не проблема.

AnatolSher, 2023-02-07:
А мне все равно :) Для 806 это не принципиально, на 801 я резистор R7 удалил. Да и прошивки можно через CKLink-Lite заливать.

board707, 2023-02-08:
Картинку добавил, но тему пока не закрываем.

AnatolSher, 2023-02-08:
Ок
Кстати, я Nictrace не много взбудоражил :) Может и он к нашей движухе присоединится

board707, 2023-02-08:
Проверил PWM пины по даташиту - есть незначительные расхождения, но в целом все правильно.  Закрываем?

AnatolSher, 2023-02-08:
Да

board707, 2023-02-08:
Я пока только ШИМ пины посмотрел. На пине PB8 на картинке не обозначен вход BRK

closed 2023-04-26 by board707