board707/w80x_arduino #13: gcc errror (solution)

icelon, 2023-04-18:
Like many of us I have found problems with the gcc compiler in Linux, I don't know if you are aware of this new version, that's why I upload it, I would have to make a tar.gz to replace the previous version

icelon, 2023-04-18:
Sorry, it does not allow uploading more than 25Mb

board707, 2023-04-23:
Please describe your problem in detail

icelon, 2023-04-27:
the version of the compiler that you use does not work on some linuxes, I wanted to ask if you can make the latest version of the tar, I tried to upload it but it limits me to 25 megabytes

board707, 2023-04-27:
What is a size of the achive?

icelon, 2023-04-29:
85 megas

board707, 2023-04-30:
Could you please upload it to any network drive, for example GoogleDisk, and send me a link?

By the way, do you have an account on the Chinese Baidu website? There is a lot of interesting information about these controllers, but registration is only by Chinese phone number.

board707, 2023-05-06:
Solved.

If you experienced the error at the first build under linux:

```
/home/user/.arduino15/packages/w80x_duino/tools/w80x_tool/v1.0/lin/wm_tool: /lib64/libc.so.6: version `GLIBC_2.34' not found (required by /home/user/.arduino15/packages/w80x_duino/tools/w80x_tool/v1.0/lin//wm_tool)
```

or something like it, just go to your '.arduino15/packages/w80x_duino/tools/w80x_tool/v1.0/lin/` folder and remove the `wm_tool` and `wm_toold` files.
After that start building again and now the error should gone.

closed 2023-05-06 by board707