board707/w80x_arduino #17: Is it possible to add i2s support on arduino?

coinight, 2023-06-18:
https://github.com/nyh-workshop/w806-i2s. Here is an example,but i have no ideal to library porting to arduino.
p.s. thanks for your hard work.

AnatolSher, 2023-06-18:
Yes, little bit later
Now I'm investigating the possibility of making SDR radio :)

board707, 2023-07-27:
@coinight 
Hi, I am starting to figure out i2s on w80x controllers. But I need an assistance from somebody, who have used i2s.
As I never dealing with it before, I need an example to test i2s on controller.
Do you have a library which you want to port to w80x or code that you require to run on our mcu ?
WBR, board707

coinight, 2023-07-27:
Hi, board707.
Thanks for your work,but i haven't use i2s on w80x.However,I used i2s on esp32 with MAX98357A to play some sounds.The link below is a guide to drive wm8978 on esp32,
ESP32 - Intro to I2S Part 1 - XTronical
Maybe the interface w80xduino need to have first is:
i2s_driver_install(I2S_HandleTypeDef  *i2s_config, int queue_size, void *i2s_queue)
i2s_driver_uninstall()
i2s_write_bytes(const void *src, size_t size)

int i2s_read_bytes(void *dest, size_t size)

the only example on w806 is nyh-workshop/w806-i2s: Simple i2s and DMA demo app for W806 (github.com) whith drives a PCM5102(also,a sound chip).
Sorry I can’t be of much help,wish your work can go smoothly.
WBR, coinight

board707, 2023-07-27:
> Maybe the interface w80xduino need to have first is:
```
i2s_driver_install(I2S_HandleTypeDef& *i2s_config, int queue_size, void *i2s_queue)
i2s_driver_uninstall()
i2s_write_bytes(const void *src, size_t size)
int i2s_read_bytes(void *dest, size_t size)
```

No, the code is not a problem.
I need some examples to test this on real hardware.
Thanks for mentioning modules (MAX98357A, PCM5102) - things are a bit clearer now. I try to find such a module somewhere and test it on it.

board707, 2023-08-09:
Hi @coinight 
I just upload to the repo a simple example sketch that playing a WAV files on W80x boards using i2s and DMA.
See the `/libraries/I2S/examples/Wav_SD_Player` folder.
The code is a port of similar ESP32 player from this link:
https://www.xtronical.com/i2s-ep3/

board707, 2023-09-26:
Hi @coinight 
Did you test the i2s support in w80x core?

closed 2023-10-15 by board707