board707/w80x_arduino #18: Cannot be found in the Development Board Manager in Arduino IDE 2.x

HalfSweet, 2023-06-26:
I tried to use it in Arduino IDE 2.x. I've added the corresponding json file, but I can't zhao'dao it in the board manager.
There is another problem, when I first installed it, the 1.x IDE would complain about a CRC checksum error. Both of these problems seem to be the result of you not adding checksum and size parameters to the json description file.
[image]

board707, 2023-06-26:
Thank you for letting us know.
I just added checksums and sizes to both json files. 

Could you please test the installation again. I'm waiting for your feedback whether the problem was solved.

HalfSweet, 2023-06-27:
I'm sorry, but the SHA256 checksum and size of the file you provided is not correct
[image]
[image]
I think you might need an automated tool to help you generate these pesky configuration files, for example, you can refer to the<https://github.com/Air-duino/Arduino-pack-json-ci>

board707, 2023-06-27:
Fixed, please test again.
I'm sorry for this mistake

HalfSweet, 2023-06-27:
Thank you very much, the CRC problem in Arduino IDE 1.x has been solved, but I still cannot install it in Arduino IDE 2.x. The IDE complains that there is no unique root directory in the archive.
```
正在下载软件包
w80x_duino:csky@2021.04.23
w80x_duino:w80x_tool@v1.0
w80x_duino:XT804@0.0.3
正在安装 w80x_duino:csky@2021.04.23
Failed to install platform: w80x_duino:XT804.
Error: 13 INTERNAL: 无法安装 w80x_duino:csky@2021.04.23 工具 : 正在搜索软件包根目录：存档中没有唯一的根目录，找到了 ‘C:\Users\HalfSweet\AppData\Local\Arduino15\tmp\package-3309983088\bin’ 和 ‘C:\Users\HalfSweet\AppData\Local\Arduino15\tmp\package-3309983088\csky-elfabiv2’

```

I think the problem may be in the toolchain, which is not in a separate directory, so there are multiple directories after unpacking
[image]
In contrast, if we use gcc's toolchain, we will find that all the other directories exist in the root directory, and perhaps for us we need to do the same
[image]
Also, I am very interested in the development of this project, can I join you to work on it? As you can see, I am also working on the <https://github.com/Air-duino> project above.

board707, 2023-06-27:
Thank you
I haven't tested arduino 2.x yet, so this problem is new to me. I need time to sort out the issue.

As about participation in the project - we will be glad to any help. Report bugs in issues, suggest improvements via PR, participate in discussions. We are very interested in any help in this project

HalfSweet, 2023-06-27:
Fine, I found ways to get it to work properly.
Like mentioned above, we just need to put a root layer on csky's tools, but I don't have publishing access, you can try to update the toolchain

board707, 2023-06-27:
Are you mean just add a "covering" folder around all contents of csky-elfabiv2-tools-mingw-minilibc-20210423.tar.gz ?
Is it related to csky package only or to w80x_tool too?

AnatolSher, 2023-06-27:
Hello! As a workaround, I installed the package in Arduino 1.19. After that, in Arduino 2.x, the package is available and working. I have two versions of the Arduino IDE installed on my computer.

board707, 2023-06-27:
> we just need to put a root layer on csky's tools,

Hi @HalfSweet 

I have repacked csky tool tarballs and add a new archives to a new json file https://github.com/board707/w80x_arduino/blob/hal-v0.6.0/package_w80x_test_index.json
Please test Arduino 2.x installation with the new package file.
Regards.

Chenjiek, 2023-08-13:
我是中国的，我是个菜鸟，我无法安装w80x显示Failed to install platform: w80x_duino:XT804.
Error: 13 INTERNAL: 无法安装 w80x_duino:csky@2021.04.23 工具 : 正在搜索软件包根目录：存档中没有唯一的根目录，找到了 ‘C:\Users\Administrator\AppData\Local\Arduino15\tmp\package-3558494723\bin’ 和 ‘C:\Users\Administrator\AppData\Local\Arduino15\tmp\package-3558494723\csky-elfabiv2’
谁能帮助我，感激不尽！

board707, 2023-08-14:
Answered in #24

closed 2023-08-14 by board707

cnwenew, 2023-08-20:
> 我是中国的，我是个菜鸟，我无法安装w80x显示无法安装平台： w80x_duino：XT804.错误： 13 内部： 无法安装 w80x_duino：csky@2021.04.23 工具 ： 正在搜索软件包根目录：存档中没有唯一的根目录，找到了 'C：\Users\Administrator\AppData\Local\Arduino15\tmp\package-3558494723\bin' 和 'C：\Users\Administrator\AppData\Local\Arduino15\tmp\package-3558494723\csky-elfabiv2' 谁能帮助我，不感激尽！

国内镜像地址： http://dl.isme.fun/w80x_arduino/package_w80x_index.json，在国内你需要正确添加开发板管理地址
文件-首选项-开发板管理器地址
（For Chinese mainland use only）

board707, 2023-08-21:
Hi
Please use a new json file https://raw.githubusercontent.com/board707/w80x_arduino/hal-v0.6.0/package_w80x_test_index.json
As was stated in #24, I can't add a correct link to the Chinese mirror, please ask a mirror maintainer.

Chenjiek, 2023-08-21:
ok