board707/w80x_arduino #21: Servo

AnatolSher, 2023-07-17:
Привет! Чтобы не забыть добавить пример в папку Basic examples
[Servo.zip](https://github.com/board707/w80x_arduino/files/12071675/Servo.zip)

board707, 2023-07-18:
Ага, добавил, отличный примерчик.

Решил проверить другие примеры на предмет ошибок.
Наткнулся на то что у нас аналоговые входы обозначены так

#define A0	PA1
#define A1	PA4
#define A2	PA3
#define A4	PA2

Мало того что не соответвуют номерам пинов, так еще и в другом порядке и вообще жуть - почему то нет А3, после А2 сразу А4 :)
Я, помнится, уже предлагал их поправить... может это я так "поправил" ?

Предлагаю сделать виртуальные номера пинов совпадающими с номером в порте А. То есть

#define A1	PA1
#define A2	PA2
#define A3     PA3
#define A4	PA4

Но при таком раскладе у нас не будет аналогового порта А0. Чтобы юзеры не путались, можно просто сделать его синонимом А1, чтобы и А0 и А1 указывали на РА1

board707, 2023-07-18:
Added