board707/w80x_arduino #22: attachInterrupt error

AnatolSher, 2023-08-03:
Привет! Тестировщики из поднебесной нарыли ошибку.
Не верна последовательность параметров передаваемых в функцию attachInterrupt(), и нет заглушки для функции digitalPinToInterrupt(). Механизм присоединения пользовательской обработки прерывания на GPIO не соответствует текущей документации на Arduino

board707, 2023-08-03:
А подробнее?

AnatolSher, 2023-08-03:
Это в документации 
Syntax
attachInterrupt(digitalPinToInterrupt(pin), ISR, mode) (recommended)
attachInterrupt(interrupt, ISR, mode) (not recommended)
attachInterrupt(pin, ISR, mode) (Not recommended. Additionally, this syntax only works on Arduino SAMD Boards, Uno WiFi Rev2, Due, and 101.)
А это в нашем примере gpio interrupt
  **attachInterrupt(PB10, FALLING, &_irq_routine);**

board707, 2023-08-03:
Понятно.

board707, 2023-08-07:
Fixed in [2eb83fc](https://github.com/board707/w80x_arduino/commit/2eb83fc0044150ae19bcac6b59d7cd18be19eda8)

closed 2023-08-07 by board707