board707/w80x_arduino #24: Failed to install platform: w80x_duino:XT804.

comrade-of-yours, 2023-08-06:
when i install this package it turns to 
Error: 13 INTERNAL: can not install w80x_duino:csky@2021.04.23 tool: Searching the package root directory：There is no unique root directory in the archive，find
/ ‘C:\Users\JerdeePC\AppData\Local\Arduino15\tmp\package-727441305\bin’ 和 ‘C:\Users\JerdeePC\AppData\Local\Arduino15\tmp\package-727441305\csky-elfabiv2’

board707, 2023-08-06:
Hi
I have repacked csky tool tarballs and add a new archives to a new json file https://github.com/board707/w80x_arduino/blob/hal-v0.6.0/package_w80x_test_index.json
Please test Arduino installation with this package file and inform me is the problem solved.
Regards.

Chenjiek, 2023-08-12:
I am in China and I am unable to use the link you posted above. I am able to use it----- http://dl.isme.fun/w80x_arduino/package_w80x_index.json----- Found in Arduino

The development board for w80x, but it cannot be installed. The error is the same as the previous friend. Can you help solve some of it? Thank you!
Error: 13 INTERNAL: can not install w80x_duino:csky@2021.04.23 tool: Searching the package root directory：There is no unique root directory in the archive，find
/ ‘C:\Users\JerdeePC\AppData\Local\Arduino15\tmp\package-727441305\bin’ 和 ‘C:\Users\JerdeePC\AppData\Local\Arduino15\tmp\package-727441305\csky-elfabiv2’

board707, 2023-08-13:
Hi, @Chenjiek 
In order to help you, somebody need to move the content of the link https://github.com/board707/w80x_arduino/blob/hal-v0.6.0/package_w80x_test_index.json to the Chinese mirror http://dl.isme.fun/w80x_arduino
Unfortunately, I do not manage the Chinese copy of this repo and cannot do so. Contact http://dl.isme.fun/w80x_arduino repository maintainer.
Please let us know if you manage to solve this problem

Chenjiek, 2023-08-13:
Unfortunately, I am unable to contact the repository administrator. I am grateful for what you have done and hope to use Arduino on w801 one day in the future。

Chenjiek, 2023-08-14:
正在下载软件包
w80x_duino:csky@2021.04.23
w80x_duino:w80x_tool@v1.0
w80x_duino:XT804@0.0.5
正在安装 w80x_duino:csky@2021.04.23
Failed to install platform: w80x_duino:XT804.
Error: 13 INTERNAL: 无法安装 w80x_duino:csky@2021.04.23 工具 : 正在搜索软件包根目录：存档中没有唯一的根目录，找到了 ‘C:\Users\Administrator\AppData\Local\Arduino15\tmp\package-1747440515\bin’ 和 ‘C:\Users\Administrator\AppData\Local\Arduino15\tmp\package-1747440515\csky-elfabiv2’
请问有谁能告诉我这个问题如何解决？

board707, 2023-08-14:
Which json file did you used for install? - 
1: https://github.com/board707/w80x_arduino/blob/hal-v0.6.0/package_w80x_index.json
or
2: https://github.com/board707/w80x_arduino/blob/hal-v0.6.0/package_w80x_test_index.json
??

Chenjiek, 2023-08-14:
I will fill in the two websites above you: https://github.com/board707/w80x_arduino/blob/hal-v0.6.0/package_w80x_index.json

Or

2: https://github.com/board707/w80x_arduino/blob/hal-v0.6.0/package_w80x_test_index.json The w80x development board is not displayed in the development board manager, so it is not displayed. I will input the information on your homepage--------- https://raw.githubusercontent.com/board707/w80x_arduino/hal-v0.6.0/package_w80x_index.json----------- This website can display a w80x_ Arduino development board, I clicked on install and encountered error 13

Chenjiek, 2023-08-14:
1: https://github.com/board707/w80x_arduino/blob/hal-v0.6.0/package_w80x_index.json
or
2: https://github.com/board707/w80x_arduino/blob/hal-v0.6.0/package_w80x_test_index.json
There is no w80x development board present

Chenjiek, 2023-08-14:
https://raw.githubusercontent.com/board707/w80x_arduino/hal-v0.6.0/package_w80x_index.json
There is a w80x development board,
正在下载软件包
w80x_duino:csky@2021.04.23
w80x_duino:w80x_tool@v1.0
w80x_duino:XT804@0.0.5
正在安装 w80x_duino:csky@2021.04.23
Failed to install platform: w80x_duino:XT804.
Error: 13 INTERNAL: 无法安装 w80x_duino:csky@2021.04.23 工具 : 正在搜索软件包根目录：存档中没有唯一的根目录，找到了 ‘C:\Users\Administrator\AppData\Local\Arduino15\tmp\package-2573158187\bin’ 和 ‘C:\Users\Administrator\AppData\Local\Arduino15\tmp\package-2573158187\csky-elfabiv2’

board707, 2023-08-14:
Hi
Make sure that you download the json file as raw - 
Try this link:
https://raw.githubusercontent.com/board707/w80x_arduino/hal-v0.6.0/package_w80x_test_index.json

If the installation will fail again - save the file from the link above to the local disk, rename it to `package_w80x_index.json` and copy the local link to the arduino board manager.
The link should be something like `file:///home/user/package_w80x_index.json`

Chenjiek, 2023-08-14:
https://raw.githubusercontent.com/board707/w80x_arduino/hal-v0.6.0/package_w80x_test_index.json
Using this address, installation is now possible. Thank you very much! It is a great honor to receive your help in a foreign country!

board707, 2023-08-14:
My congratulations!

closed 2023-08-14 by board707

guduxue0624, 2023-12-28:
csky-elfabiv2-gcc: error: Files\Arduino15\packages\w80x_duino\hardware\XT804\0.0.5\cores\w806/lib/libdsp.a: No such file or directory
csky-elfabiv2-gcc: error: Files\Arduino15\packages\w80x_duino\hardware\XT804\0.0.5\cores\w806/ld/gcc_csky.ld: No such file or directory

exit status 1

Compilation error: exit status 1

board707, 2023-12-28:
Hi
Please open a new issue about your problem