board707/w80x_arduino #25: SPI issues

RudolphRiedel, 2023-08-06:
I am just playing with a W806 board and SPI.
I intend to look deeper but this is what I found now.

I have this little sketch:
#include <Arduino.h>
#include <SPI.h>

#define EVE_CS PB10
#define EVE_PDN PB11

void setup()
{
  pinMode(EVE_CS, OUTPUT);
  digitalWrite(EVE_CS, HIGH);
  pinMode(EVE_PDN, OUTPUT);
  digitalWrite(EVE_PDN, LOW);

  SPI.begin(); /* sets up the SPI to run in Mode 0 and 1 MHz */
  SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0));
}

void loop()
{
  delay(2);

  digitalWrite(EVE_CS, LOW);
  SPI.transfer(0x55);
  SPI.transfer(0xaa);
  SPI.transfer(0x55);
  digitalWrite(EVE_CS, HIGH);

  digitalWrite(EVE_CS, LOW);
  uint32_t buffer = 0x55aa55aa;
  SPI.transfer(&buffer, 3);
  digitalWrite(EVE_CS, HIGH);
}

And this is what I get in the logic analyzer:
[image]

So well, on a quick glance everything is working, there are a couple of issues though.
- the frequency of that SPI is not 8 MHz, it is 1MHz.
- the pause between the single byte transfers is over 7µs
- MOSI behaves a bit odd, like the default seems to be low but sometimes it goes high

[image]
[image]
[image]

I'll have a look to see what can be done about this.

AnatolSher, 2023-08-06:
Hello
[image]
Default SS pin
[image]
Doc - https://github.com/board707/w80x_arduino/blob/hal-v0.6.0/doc/SPI.md
When we select 20MHz for SPI, it means base peripheral clock frequency is 40mHz/prescaler
[image]
As a result, we have a max spi clock frequency of 20 Mhz

RudolphRiedel, 2023-08-06:
Thanks for pointing out that SPI.md

Yes, I just worked thru SPI.cpp, reformatted it to a more consistence style and translated the few russian comments.
And while it is still working I noticed the limitations and switched to 10MHz which is working.

The pauses got a bit shorter with the increased SPI clock. 

[image]

I will need a software controlled CS, so I just continue to use it.

board707, 2023-08-06:
> I noticed the limitations and switched to 10MHz which is working.

Yes, the package probably lacks clear style and detailed documentation.
However, as far as I understand, the problem is generally solved?

RudolphRiedel, 2023-08-06:
Nah, it's still rather slow, especially for a 240MHz core.
But that may be something to fix here:
https://github.com/IOsetting/wm-sdk-w806/blob/main/platform/drivers/wm_spi.c
Well, or with register based code in SPI.cpp.

I'll have a look, but not right away.

board707, 2023-08-06:
> But that may be something to fix here:
> https://github.com/IOsetting/wm-sdk-w806/blob/main/platform/drivers/wm_spi.c

Our code based on the exact this file.
But anyway, it will be usefull if look on it again