board707/w80x_arduino #30: Conversion wait of 1ms is ridiculous - can we set up a continuous ADC conversion on Callback ???

jlsilicon, 2023-10-11:
A Conversion wait of 1ms on this high speed microcontroller seems ridiculous.

Could we have a continuous loop reading the ADC pin(s) - then post the readings in a global Var to read at our convenience - instead of holding the micro up ?

Ie: Is there a way simple to Start the Adc Conversion, then Set the Result into a glabal (such as LastAdcReading) in the Callback.

=>

Looking at : 

int HAL_ADC_GET_INPUT_VOLTAGE(ADC_HandleTypeDef* hadc)
{
    int value;
    double voltage = 0.0;
    
    assert_param(IS_ADC_ALL_INSTANCE(hadc->Instance));
    
    HAL_ADC_Start(hadc);
    HAL_ADC_PollForConversion(hadc);
    HAL_ADC_Stop(hadc);

    value = HAL_ADC_GetValue(hadc);
    voltage = (double)value / 4.0 * (126363 / 1000.0) / 1000000 + 1.196;
    
    value = (int)(voltage * 1000);
    return value;
}

=>

For example to use :

double  ADC_Last = 0.0 ;

main()
{
...
  ADC_Init();
  // HAL_ADC_Init(hadc);
...
}

static void ADC_Init(void)
{
    hadc.Instance     = ADC;
    hadc.Init.channel = ADC_CHANNEL_0;
    hadc.Init.freq    = 1000;
    
    if (HAL_ADC_Init(&hadc) != HAL_OK)
    {
        Error_Handler();
    }
}

void ADC_Start()
{
    assert_param(IS_ADC_ALL_INSTANCE(hadc->Instance));
    
    HAL_ADC_Start(hadc);
//    HAL_ADC_PollForConversion(hadc);
}

void HAL_ADC_ConvCpltCallback(ADC_HandleTypeDef* hadc)
{
    int value;
    double voltage = 0.0;
    
//    HAL_ADC_PollForConversion(hadc);
    HAL_ADC_Stop(hadc);

    value = HAL_ADC_GetValue(hadc);
    voltage = (double)value / 4.0 * (126363 / 1000.0) / 1000000 + 1.196;
    
    value = (int)(voltage * 1000);
    ADC_Last = value ;

    ;

    assert_param(IS_ADC_ALL_INSTANCE(hadc->Instance));
    
    HAL_ADC_Start(hadc);
//    HAL_ADC_PollForConversion(hadc);
}

AnatolSher, 2023-10-11:
Good question... We need to think about it :)

board707, 2023-10-15:
> A Conversion wait of 1ms on this high speed microcontroller seems ridiculous.

I could agree with you, but datasheet clearly stated, that the maximum ADC rate is 1000 samples in second - so give us a 1ms delay between samples.

closed 2023-10-22 by board707