board707/w80x_arduino #31: Can Not get PSRAM_test.ino to work - using Ips6404-SQPI

jlsilicon, 2023-10-13:
Can Not get PSRAM_test.ino to work - using Ips6404-SQPI

I integrated the Ips6404-SQPI to W806 :
- but it only returns and prints garbage back.

/*
  Pad pin mapping relationship
SCLK    -> PA_15 PSRAM_CK 
SI/IO0  -> PB_02 PSRAM_D0 
01/SO   -> PB_03 PSRAM_D1 
WP      -> PB_04 PSRAM_D2 
HOLD    -> PB_05 PSRAM_D3 
CS      -> PB_27 PSRAM_CS 

        BRD PSRAM :
PB27   CS  \/  +Vdd
PB3 01/SO      Hold  PB5 
PB4    WP      SCLK  PA15  
     Gnd-    SI/IO0  PB2 
*/

board707, 2023-10-15:
Hi 
This module has been added by @AnatolSher.
I will ask him to review the issue.

AnatolSher, 2023-10-15:
Hello! I had testing PSRAM sketch with Air103 board and PS6404L-3SQR-SN chip. It works well.
Maybe your chips are fake? I was sent defective chips from Aliexpress.
[image]

jcoleman-oneaihealth, 2023-10-16:
My Chips work - on here if I just use BitBang Code

jcoleman-oneaihealth, 2023-10-16:
I also can not get the SD Card examples - to connect to the (working) SD card.

I thought that I had this working  6 months ago , but now.
Possibly I am using the wrong Arduino IDE / Libraries since Arduino updates ?
Arduino 1.8.19

AnatolSher, 2023-10-16:
It's hard for me to imagine what's happening on your computer...
But it works well
[image]

AnatolSher, 2023-10-18:
Hello! For w801 init function must be
static void PSRAM_Init(void)
{
  __HAL_RCC_PSRAM_CLK_ENABLE();
  __HAL_RCC_GPIO_CLK_ENABLE();
  // For W801
  __HAL_AFIO_REMAP_PSRAM_CS(GPIOB, GPIO_PIN_27);
  __HAL_AFIO_REMAP_PSRAM_CLK(GPIOB, GPIO_PIN_0);
  __HAL_AFIO_REMAP_PSRAM_D0(GPIOB, GPIO_PIN_2);
  __HAL_AFIO_REMAP_PSRAM_D1(GPIOB, GPIO_PIN_3);
  __HAL_AFIO_REMAP_PSRAM_D2(GPIOB, GPIO_PIN_4);
  __HAL_AFIO_REMAP_PSRAM_D3(GPIOB, GPIO_PIN_5);

  hpsram.Instance = PSRAM;
  hpsram.Init.Div = 2;
  hpsram.Init.Mode = PSRAM_MODE_QSPI;

  if (HAL_PSRAM_Init(&hpsram) != HAL_OK)
  {
    Serial.printf("Init error...\r\n");
  }
}
It is work!

jcoleman-oneaihealth, 2023-10-18:
Sorry, I followed that code above, except I used : 
`
  __HAL_AFIO_REMAP_PSRAM_CLK(GPIOA, GPIO_PIN_15);
//  __HAL_AFIO_REMAP_PSRAM_CLK(GPIOB, GPIO_PIN_0);
`
I used chip :  Ips6404L-SQ .
The specs seem to match to your chip.
But the data Written - Reads back as garbage chars.

AnatolSher, 2023-10-18:
Yes, I see this too. It doesn't work well on W801 :)
But Air103 is the same, only there is no RF

jcoleman-oneaihealth, 2023-10-18:
I am using W806 - is this code compatible with it ?

AnatolSher, 2023-10-18:
Must be
I plan to test this on the w806 tomorrow

jcoleman-oneaihealth, 2023-10-18:
In your code , I do not see the define for :

  PSRAM_ADDR_START

jcoleman-oneaihealth, 2023-10-18:
Your UserMain() code does not compile on the Board707 code.
What are you compiling with ?

AnatolSher, 2023-10-19:
https://github.com/ganhailin/wm_sdk_w801 and CDK dev inv

AnatolSher, 2023-10-19:
Hello. Good news for you :) It work on W806
[image]
You need this:
[image]
Result:
[image]

jcoleman-oneaihealth, 2023-10-19:
Can you compile with the ? :
https://github.com/board707/w80x_arduino

jcoleman-oneaihealth, 2023-10-19:
So what was the code for board707 compile ?

AnatolSher, 2023-10-19:
Copy and paste to arduino ide
#include "Arduino.h"

// Tested on w806 board with PS6404L-3SQR-SN chip

PSRAM_HandleTypeDef hpsram;

static void PSRAM_Init(void);

static uint8_t *psram_buf = (uint8_t *)PSRAM_ADDR_START;

void setup()
{
  Serial.begin(115200);
  PSRAM_Init();
}
 
 
void loop()
{
  uint8_t temp[100];
  int i = 0;

  memset(psram_buf, 0, 100);
  for(i = 0; i < 100; i++)
  {
    Serial.printf("%x ", psram_buf[i]);
  }
  Serial.printf("\r\n");

  for(i = 0; i < 100; i++)
  {
    temp[i] = i % 256;
  }

  memcpy(psram_buf, temp, 100);
  for(i = 0; i < 100; i++)
  {
    Serial.printf("%x ", psram_buf[i]);
  }
  Serial.printf("\r\n");

  delay(1000);
}

static void PSRAM_Init(void)
{
  __HAL_RCC_PSRAM_CLK_ENABLE();
  __HAL_RCC_GPIO_CLK_ENABLE();
  __HAL_AFIO_REMAP_PSRAM_CS(GPIOB, GPIO_PIN_1);
  __HAL_AFIO_REMAP_PSRAM_CLK(GPIOB, GPIO_PIN_0);
  __HAL_AFIO_REMAP_PSRAM_D0(GPIOB, GPIO_PIN_2);
  __HAL_AFIO_REMAP_PSRAM_D1(GPIOB, GPIO_PIN_3);
  __HAL_AFIO_REMAP_PSRAM_D2(GPIOB, GPIO_PIN_4);
  __HAL_AFIO_REMAP_PSRAM_D3(GPIOB, GPIO_PIN_5);

  hpsram.Instance = PSRAM;
  hpsram.Init.Div = 3;
  hpsram.Init.Mode = PSRAM_MODE_QSPI;

  if (HAL_PSRAM_Init(&hpsram) != HAL_OK)
  {
    Serial.printf("Init error...\r\n");
  }
}
Connect the chip correctly ! CS->PB1;CLK->PB0
Enjoy :)

jcoleman-oneaihealth, 2023-10-19:
/*

COMPILES 

-

Copy and paste to arduino ide

Connect the chip correctly ! 
  CS->PB1;
  CLK->PB0

// Tested on w806 board with PS6404L-3SQR-SN chip

-

  BRD PSRAM :
PB1   CS   \/  +Vdd
PB3  IO1/SO IO3/Hold    PB5 
PB4  IO2/WP     SCLK    PB0  
      Gnd-    SI/IO0    PB2 

       Ly68L6400sLi :
    CE!     \/   +Vcc
    SO/SIO1      SIO3
    SIO2         SCLK
    -Vss      SI/SIO0

*/

//

#include "Arduino.h"

//

PSRAM_HandleTypeDef hpsram;

static void PSRAM_Init(void);

static uint8_t *psram_buf = (uint8_t *)PSRAM_ADDR_START;

//

void setup()
{
  Serial.begin(115200);
  Serial.printf("\r\n> : W806_PSRAM_test_3 : \r\n\r\n");

  PSRAM_Init();
}

void loop()
{
uint8_t temp[100];
int i = 0;

  Serial.printf("- Clear & Read PSRAM : \r\n");
  memset(psram_buf, 0, 100);
  for(i = 0; i < 100; i++)
  {
    Serial.printf("%x ", psram_buf[i]);
  }
  Serial.printf("\r\n");

  Serial.printf("- Write Inc# & Read PSRAM : \r\n");
  for(i = 0; i < 100; i++)
  {
    temp[i] = i % 256;
  }

  Serial.printf("- Clear & Read PSRAM : \r\n");
  memcpy(psram_buf, temp, 100);
  for(i = 0; i < 100; i++)
  {
    Serial.printf("%x ", psram_buf[i]);
  }
  Serial.printf("\r\n");

  Serial.printf("- Write Letters & Read PSRAM : \r\n");
    Serial.printf(  "- W: #0=[A] ");
     psram_buf[0] = 'A' ;
     Serial.printf(  "R: #0=>[%c] ",psram_buf[0]);

    Serial.printf("; - W: #1=[A] ");
     psram_buf[1] = 'z' ;
     Serial.printf(", R: #1=>[%c] ",psram_buf[1]);

     Serial.printf("; R: #0=>[%c] ",psram_buf[0]);
     Serial.printf(", R: #1=>[%c] ",psram_buf[1]);
  Serial.printf("\r\n");

  Serial.printf("\r\n");

delay(1000);
}

//

static void PSRAM_Init(void)
{
  __HAL_RCC_PSRAM_CLK_ENABLE();
  __HAL_RCC_GPIO_CLK_ENABLE();
  __HAL_AFIO_REMAP_PSRAM_CS(GPIOB, GPIO_PIN_1);
  __HAL_AFIO_REMAP_PSRAM_CLK(GPIOB, GPIO_PIN_0);
  __HAL_AFIO_REMAP_PSRAM_D0(GPIOB, GPIO_PIN_2);
  __HAL_AFIO_REMAP_PSRAM_D1(GPIOB, GPIO_PIN_3);
  __HAL_AFIO_REMAP_PSRAM_D2(GPIOB, GPIO_PIN_4);
  __HAL_AFIO_REMAP_PSRAM_D3(GPIOB, GPIO_PIN_5);

  hpsram.Instance = PSRAM;
  hpsram.Init.Div = 3;
  hpsram.Init.Mode = PSRAM_MODE_QSPI;

  if (HAL_PSRAM_Init(&hpsram) != HAL_OK)
  {
    Serial.printf("Init error...\r\n");
  }
}

///

jcoleman-oneaihealth, 2023-10-19:
OUTPUT : 

- Clear & Read PSRAM : 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
- Write Inc# & Read PSRAM : 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
- Clear & Read PSRAM : 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 60 61 62 63   // <- Old Data ??? 
- Write Letters & Read PSRAM : 
- W: #0=[A] 
   R: #0=>[A] ; 

- W: #1=[A] , 
   R: #1=>[z] ; 

- R: #0=>[] ,    // <-  The data is last char Written , but not Reading / Writing  PSRAM => char z ???
- R: #1=>[z]

-

The data is last char Written , but not Reading PSRAM

jcoleman-oneaihealth, 2023-10-19:
Does your PSRAM code include ? :

//   WrtByt: RESET_EN = 0x66 = 0b01100110 :  //
  SS_L;  
    SPI_WrtByt( 0x66 );
  SS_H;  

//   WrtByt: RESET = 0x99 = 0b10011001 :  //
  SS_L;  
    SPI_WrtByt( 0x99 );
  SS_H;

jcoleman-oneaihealth, 2023-10-19:
My chip seems to be Reading data, but Not Writing Data to PSRAM.

AnatolSher, 2023-10-19:
There is a specialized SPRAM controller on board the w806 chip. Thanks to it, we can address it like regular memory. SPI interface is not needed here. Show me a photo of how you connected the psram chip and the board

AnatolSher, 2023-10-19:
Initialization of the psram controller is described in the HAL. You can find this in the drivers folder C:\Users\User\AppData\Local\Arduino15\packages\w80x_duino\hardware\XT804\0.0.5\cores\w806\lib\drivers

jcoleman-oneaihealth, 2023-10-19:
I think its wired correctly - if I keep reading ABC at address 96.

[image]

jcoleman-oneaihealth, 2023-10-19:
OUTPUT :

- Clear & Read PSRAM : 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
60 ` 61 a 62 b 63 c

AnatolSher, 2023-10-19:
you are reading the latest data from the controller's psram buffer. not from a chip

jcoleman-oneaihealth, 2023-10-19:
strcpy((char*)psram_buf, (char*)"AbCdEfGhIjKlMnOpQrStUvWxYz *** ");  
=>
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 2A * 2A * 20   00 . 

//  strcpy((char*)psram_buf, (char*)"0 1 2 3 4 5 6 7 8 9 *** ");
=>
00 . 00 . 00 . 7A z 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 .

jcoleman-oneaihealth, 2023-10-19:
I recompiled at 80MHz - no difference.

AnatolSher, 2023-10-19:
hpsram.Init.Div = 32; - no; from 2 to 7

AnatolSher, 2023-10-19:
You need to solder the chip on top of the board. psram is a very capricious thing. There may be bad contacts on the blogboard

AnatolSher, 2023-10-19:
> Does your PSRAM code include ? :
> 
> // WrtByt: RESET_EN = 0x66 = 0b01100110 : // digitalWrite( PB1 , LOW ); SPI.transfer( 0x66 ); digitalWrite( PB1 , HIGH );
> 
> // WrtByt: RESET = 0x99 = 0b10011001 : // digitalWrite( PB1 , LOW ); SPI.transfer( 0x99 ); digitalWrite( PB1 , HIGH );
> 
> -- Needed after PSRAM powerup , before Writing data

When we to do soft reset the psram controller, this sequence is transmitted automatically.
PSRAM reset
Writing 1 to control register starts the reset operation to PSRAM, and it is automatically cleared after reset.

jcoleman-oneaihealth, 2023-10-19:
I see, I am stuck with just BitBang or SPI.tx/rx .

Let me try the SPI.tx/rx then...

jcoleman-oneaihealth, 2023-10-21:
OK.
Figured out part of the problem.
I had PSRAM +Vcc connected to Gnd.

jcoleman-oneaihealth, 2023-10-21:
Now I get this : 

* Writing "AZaz " : 
* Reading : 
 => "⸮⸮⸮⸮⸮" : 

- Read Default PSRAM : 
41 A 71 q 75 u 77 w 56 V 66 f 55 U 68 h 99 . 66 f 4B K 64 d 4D M 66 f 4F O 55 U 
55 U 77 w 51 Q 77 w 55 U 77 w 55 U 71 q 51 Q 77 w 02 . A2 . A2 . A2 . 00 . 02 . 
02 . 12 . 22 " 32 2 42 B 52 R 62 b 72 r 82 . 92 . A2 . B2 . C2 . D2 . E2 . F3 . 
30 0 31 1 33 3 33 3 31 1 31 1 32 2 33 3 31 1 31 1 3A : 33 3 31 1 31 1 3E > B3 . 
04 . 11 . 42 B 11 . 44 D 54 T 46 F 47 G 88 . 99 . 4A J 91 . 4C L DD . 4E N F5 . 
50 P 55 U 55 U 51 Q 55 U 55 U 55 U 55 U 00 . 00 . 00 . 05 . 55 U 55 U 5E ^ 55 U 
60 ` 11 . 46 F 63 c 
- Clear & Read PSRAM : 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 00 . 
00 . 00 . 00 . 00 . 
- Write Inc# & Read PSRAM : 
00 . 11 . 02 . 33 3 04 . 55 U 06 . 51 Q 08 . 99 . 0A . 91 . 0C . D5 . 0E . 80 . 
10 . 11 . 12 . 11 . 14 . 15 . 16 . 11 . 18 . 11 . 1A . 91 . 1C . 91 . 1E . 91 . 
20   12 . 22 " 32 2 42 B 52 R 66 f 27 ' 82 . 92 . A2 . B2 . C2 . DD . 2E . F3 . 
30 0 31 1 33 3 33 3 30 0 31 1 32 2 33 3 31 1 31 1 33 3 3B ; 30 0 31 1 3E > B3 . 
40 @ 11 . 44 D 43 C 44 D 45 E 46 F 47 G 88 . 99 . 4A J 91 . 4C L D5 . 4E N D5 . 
50 P 51 Q 51 Q 51 Q 55 U 55 U 55 U 55 U 51 Q 51 Q 5A Z 51 Q 5C \ 55 U 5E ^ 55 U 
60 ` 61 a 62 b 63 c

jcoleman-oneaihealth, 2023-10-21:
Code:

`
//
#include "Arduino.h"
#include <SPI.h>
//

PSRAM_HandleTypeDef hpsram;
static void PSRAM_Init(void);
static uint8_t *psram_buf = (uint8_t *)PSRAM_ADDR_START;
uint8_t temp[100];

//

void setup()
{
//  SystemClock_Config(CPU_CLK_240M);  /// - 240MHz 

  Serial.begin(115200);
  Serial.printf("\r\n> : W806_PSRAM_test_3 : \r\n\r\n");

#define  SPI_TEST_E  1 
#ifdef   SPI_TEST_E
  SPI_PSRAM_Reset();

  Serial.printf("* Writing \"AZaz \" : \r\n");
  digitalWrite( PB1 , LOW );
   SPI.transfer( 0x02 );
    SPI.transfer( 0x00 );
    SPI.transfer( 0x00 );
    SPI.transfer( 0x00 );
   SPI.transfer( 0x41 );
   SPI.transfer( 0x61 );
   SPI.transfer( 0x5A );
   SPI.transfer( 0x7A );
   SPI.transfer( 0x20 );
  digitalWrite( PB1 , HIGH );

  Serial.printf("* Reading : \r\n");
  digitalWrite( PB1 , LOW );
   SPI.transfer( 0x03 );
    SPI.transfer( 0x00 );
    SPI.transfer( 0x00 );
    SPI.transfer( 0x00 );
   temp[0] = SPI.transfer( 0xFF );
   temp[1] = SPI.transfer( 0xFF );
   temp[2] = SPI.transfer( 0xFF );
   temp[3] = SPI.transfer( 0xFF );
   temp[4] = SPI.transfer( 0xFF );
   temp[5] = 0;
  digitalWrite( PB1 , HIGH );
  Serial.printf(" => \"%s\" : \r\n\r\n" , temp );
#endif
  
  PSRAM_Init();
//  SPI_PSRAM_Reset();
}

void loop()
{
// uint8_t temp[100];
int     i = 0;
char    c ;

  Serial.printf("- Read Default PSRAM : \r\n");
  for(i = 0; i < 100; i++)
  {
    Serial.printf("%02X ", (c=psram_buf[i]) );

    if( c > 0x1F  &&  c < 0x80 )
      Serial.printf("%c ", c );  
    else
      Serial.printf(". ");  

    if( i % 16 == 15 )
      Serial.printf("\r\n");  
  }
  Serial.printf("\r\n");

  Serial.printf("- Clear & Read PSRAM : \r\n");
  memset(psram_buf, 0, 100);
  for(i = 0; i < 100; i++)
  {
    Serial.printf("%02X ", (c=psram_buf[i]) );

    if( c > 0x1F  &&  c < 0x80 )
      Serial.printf("%c ", c );  
    else
      Serial.printf(". ");  

    if( i % 16 == 15 )
      Serial.printf("\r\n");  
  }
  Serial.printf("\r\n");

  Serial.printf("- Write Inc# & Read PSRAM : \r\n");
  for(i = 0; i < 100; i++)
  {
    temp[i] = i % 256;
  }
  memcpy(psram_buf, temp, 100);
  for(i = 0; i < 100; i++)
  {
    Serial.printf("%02X ", (c=psram_buf[i]) );

    if( c > 0x1F  &&  c < 0x80 )
      Serial.printf("%c ", c );  
    else
      Serial.printf(". ");  

    if( i % 16 == 15 )
      Serial.printf("\r\n");  
  }
  Serial.printf("\r\n");

  Serial.printf("\r\n- Set Abc & Read PSRAM : \r\n");
  strcpy((char*)psram_buf, (char*)"AbCdEfGhIjKlMnOpQrStUvWxYz *** ");
//  strcpy((char*)psram_buf, (char*)"0 1 2 3 4 5 6 7 8 9 *** ");
//  for(i = 0; i < 100; i++)
  for(i = 0; i < 128; i++)
//  for(i = 0; i < 1000; i++)
//  for(i = 0; i < 10000; i++)
  {
    Serial.printf("%02X ", (c=psram_buf[i]) );

    if( c > 0x1F  &&  c < 0x80 )
      Serial.printf("%c ", c );  
    else
      Serial.printf(". ");  

    if( i % 16 == 15 )
      Serial.printf("\r\n");  
  }
  Serial.printf("\r\n");

  Serial.printf("\r\n- Write Letters & Read PSRAM : \r\n");
    Serial.printf(  "- W: #0=[A] ");
     psram_buf[0] = 'A' ;
     Serial.printf(  ", R: #0=>[%c] \r\n",psram_buf[0]);

    Serial.printf("- W: #1=[x] ");
     psram_buf[1] = 'x' ;
     Serial.printf(", R: #1=>[%c] \r\n",psram_buf[1]);

    Serial.printf("- W: #2=[y] ");
     psram_buf[2] = 'y' ;
     Serial.printf(", R: #2=>[%c] \r\n",psram_buf[2]);

    Serial.printf("- W: #3=[z] ");
     psram_buf[3] = 'z' ;
     Serial.printf(", R: #3=>[%c] \r\n",psram_buf[3]);

    Serial.printf("=> R: #0=>[%c] \r\n",psram_buf[0]);
     Serial.printf("   R: #1=>[%c] \r\n",psram_buf[1]);
     Serial.printf("   R: #2=>[%c] \r\n",psram_buf[2]);
     Serial.printf("   R: #3=>[%c] \r\n",psram_buf[3]);

  Serial.printf("\r\n");

  delay(1000);
}

//

static void PSRAM_Init(void)
{
  __HAL_RCC_PSRAM_CLK_ENABLE();
  __HAL_RCC_GPIO_CLK_ENABLE();
  
  __HAL_AFIO_REMAP_PSRAM_CS(GPIOB, GPIO_PIN_1);   // PB1 / CS 
  __HAL_AFIO_REMAP_PSRAM_CLK(GPIOB, GPIO_PIN_0);  // PB0 / SCLK 
  __HAL_AFIO_REMAP_PSRAM_D0(GPIOB, GPIO_PIN_2);   // PB2 / MOSI 
  __HAL_AFIO_REMAP_PSRAM_D1(GPIOB, GPIO_PIN_3);   // PB3 / MISO 
  __HAL_AFIO_REMAP_PSRAM_D2(GPIOB, GPIO_PIN_4);   // PB4 
  __HAL_AFIO_REMAP_PSRAM_D3(GPIOB, GPIO_PIN_5);   // PB5 

  hpsram.Instance  = PSRAM;
 //  hpsram.Init.Div  = 3;
   hpsram.Init.Div  = 4;
//  hpsram.Init.Div  = 15;  // - 2-15
//  hpsram.Init.Div  = 20;
//  hpsram.Init.Div  = 32;
  hpsram.Init.Mode = PSRAM_MODE_QSPI;

  if (HAL_PSRAM_Init(&hpsram) != HAL_OK)
  {
    Serial.printf("Init error...\r\n");
  }

//  SPI_PSRAM_Reset();

}

void  SPI_PSRAM_Reset()
{

  /// Reset PSRAM : ///
  ///  HardSPI::transfer(uint8_t val) : ///
//  HardSPI::transfer( 0x66 );

  /// SPI.setSPIpins(uint8_t mosi, uint8_t miso, uint8_t sck) : ///
  SPI.setSPIpins(            PB2,          PB3,          PB0 );
  SPI.begin();

  pinMode( PB1 , OUTPUT );
  digitalWrite( PB1 , HIGH );

  digitalWrite( PB1 , LOW );
   SPI.transfer( 0x66 );
  digitalWrite( PB1 , HIGH );

  digitalWrite( PB1 , LOW );
   SPI.transfer( 0x99 );
  digitalWrite( PB1 , HIGH );
}
`

jcoleman-oneaihealth, 2023-10-21:
I tried with short Wires + Caps (1uF , 0.1uF).

Still same output.

jcoleman-oneaihealth, 2023-10-22:
I tried with your very short Wires.

Just got worse :

- Clear & Read PSRAM : 
EC . A9 . AE . AF . AE . FF . FF . EC . EC . A9 . AE . AF . AE . FF . FF . EC . 
89 . FF . FF . DD . FF . AE . DD . 88 . FF . A8 . FF . AF . FF . AE . FF . 88 . 
FF . 89 . FF . AA . EA . 9A . FF . 8E . DD . 89 . FF . 98 . EE . 9A . FF . 8E . 
CD . FF . FF . AA . BB . A8 . E8 . FF . CD . FF . FF . FF . BB . A8 . E8 . FF . 
8F . DE . 9B . CF . FF . FF . EE . E8 . 8F . DE . 9B . CF . FF . FF . EA . E8 . 
FF . FF . 8F . FF . FF . A9 . FF . FA . FF . AF . BD . 88 . BB . A9 . FF . FA . 
00 . 00 . 00 . 00 . 

Waiting on your PSRAM chips that I ordered : 
-- ESP-PSRAM64H 
-- LY68L6400SLIT

jcoleman-oneaihealth, 2023-10-22:
I noticed that you connected your CS to Gnd.
I matched that and just got 0xFF ...

jcoleman-oneaihealth, 2023-10-22:
Your Code , does Not match your Chip Wiring : 

`
static void PSRAM_Init(void)
{
__HAL_RCC_PSRAM_CLK_ENABLE();
__HAL_RCC_GPIO_CLK_ENABLE();
__HAL_AFIO_REMAP_PSRAM_CS(GPIOB, GPIO_PIN_1);
__HAL_AFIO_REMAP_PSRAM_CLK(GPIOB, GPIO_PIN_0);
__HAL_AFIO_REMAP_PSRAM_D0(GPIOB, GPIO_PIN_2);
__HAL_AFIO_REMAP_PSRAM_D1(GPIOB, GPIO_PIN_3);
__HAL_AFIO_REMAP_PSRAM_D2(GPIOB, GPIO_PIN_4);
__HAL_AFIO_REMAP_PSRAM_D3(GPIOB, GPIO_PIN_5);

hpsram.Instance = PSRAM;
hpsram.Init.Div = 3;
hpsram.Init.Mode = PSRAM_MODE_QSPI;
`

jcoleman-oneaihealth, 2023-10-22:
Ok , I Fixed it !
My Wiring was Wrong :
PB9 - is mislabeled as PBS - which looks like PB5 ...

Aps6404 Chip is good.

jcoleman-oneaihealth, 2023-10-22:
Everything is fine now.

Except : 
- If I malloc() after PSRAM is initialized => then PSRAM fails after this ???

AnatolSher, 2023-10-23:
Congratulations!
LY68L6400SLIT I will check when the package arrives from Aliexpress. I hope that I will be lucky and they will send me good chips. Follow the discussions, I will publish the result there. malloc() is not suitable for dynamic memory allocation. There are other approaches that WinnerMicro recommends. A little bit later I will give code examples. If you have solved your problems at this stage, please close the issue

jcoleman-oneaihealth, 2023-10-23:
I dont think Github is a problem - it does not cause a problem now.

I think that it was just too long leads - which fixed it.

jlsilicon, 2023-10-23:
> @jcoleman-oneaihealth
> If your issue has been fixed - please close the issue and star our repo! :)

You are the Owner - I can Not Close it ...