board707/w80x_arduino #33: Can Not get (2) PSRAM chips to work together

jcoleman-oneaihealth, 2023-11-08:
Hey AnatolSher , thanks for the help with gettig the 1st PSRAM chip to work !

But now I am using multiple PSRAM chips.

I stacked (2) chips.  

        BRD PSRAM :
PB6 + PB1   CS        \/    +Vdd
          PB3   IO1/SO  IO3/Hold    PB5 
          PB4   IO2/WP        SCLK    PB0  
                   Gnd-          SI/IO0    PB2 

1st  Chip has CS connected to BP1 - with 100K resistor tied to V++.
2nd Chip has CS connected to BP6 - with 100K resistor tied to V++ also.

It works fine, if I only Initialize PSRAM with CS PB1.

But if I disable PB6 cs as : 
```
  /// DeSelect cs=PB1 : ///
  pinMode(      PB1 , OUTPUT );
  digitalWrite( PB1 , HIGH   );

```
Then ReInit PSRAM : 

`  PSRAM_ReInit_CS( GPIOB , GPIO_PIN_6 );
`

```
static void  PSRAM_ReInit_CS( GPIO_TypeDef* gpio_N , int cs_N )
{
 //  __HAL_RCC_PSRAM_CLK_ENABLE();
 //  __HAL_RCC_GPIO_CLK_ENABLE();

/*  
  BRD PSRAM :
PB1   CS    \/   +Vdd
PB3   IO1/SO IO3/Hold    PB5 
PB4   IO2/WP     SCLK    PB0  
      Gnd-     SI/IO0    PB2 
*/

 //  __HAL_AFIO_REMAP_PSRAM_CS(GPIOB, GPIO_PIN_1);   // PB1 / CS   // CS 
  __HAL_AFIO_REMAP_PSRAM_CS( gpio_N , cs_N );   //  ..  CS   // CS 
 
  __HAL_AFIO_REMAP_PSRAM_CLK(GPIOB, GPIO_PIN_0);  // PB0 / SCLK // SCK 
  __HAL_AFIO_REMAP_PSRAM_D0(GPIOB, GPIO_PIN_2);   // PB2 / MOSI // SI/IO0 
  __HAL_AFIO_REMAP_PSRAM_D1(GPIOB, GPIO_PIN_3);   // PB3 / MISO // SO/IO1 
  __HAL_AFIO_REMAP_PSRAM_D2(GPIOB, GPIO_PIN_4);   // PB4        // WP/IO2 
  __HAL_AFIO_REMAP_PSRAM_D3(GPIOB, GPIO_PIN_5);   // PB5        // Hold/IO3 

  hpsram.Instance  = PSRAM;
 //   hpsram.Init.Div  = 2;    // - (160MHz) BAD DATA on Ips6404 - 
   hpsram.Init.Div  = 3;    // - (80MHz) WORKS on Ips6404 -
//  hpsram.Init.Div  = 0 ;  // => 0xFF ...
////   hpsram.Init.Div  = 4 ; // = 5 , = 8 ;
 ////  hpsram.Init.Div  = 15;  // = 2-15
////  hpsram.Init.Div  = 20;
////  hpsram.Init.Div  = 32;
  hpsram.Init.Mode = PSRAM_MODE_QSPI;

  if (HAL_PSRAM_Init(&hpsram) != HAL_OK)
  {
    Serial.printf("Init error...\r\n");
  }
}

```

It just spits out FFs ...

AnatolSher, 2023-11-09:
Hello! It doesn't work that way. You need to use an analog bidirectional switch.

jcoleman-oneaihealth, 2023-11-10:
I tried that using a 74hc4051.
But it did not work.
I was thinking maybe it was too slow.

-

So, I tried with just : 
PB1-> Ram1chip:CS
PB6-> Ram2chip:CS

That was the code above.
But it just spits out 0xFFs with PB6 as CS in ReInit()

AnatolSher, 2023-11-10:
> Whats wrong with just (2) gpio to (2) CS pins - instead of MUX ?

Because it doesn't make sense. The CS line is controlled by the PSRAM controller in hardware. We can only change the time that this line is held low level. 
The multiplexer must be selected high-frequency. SN74LVC (TI) or ADG (AD)
I don't understand why you need 2 banks, but you should simulate the standard connection of one bank

jcoleman-oneaihealth, 2023-11-11:
How about : SN74LVC138APW  ?

AnatolSher, 2023-11-11:
No. This is not the right chip for your purposes.
2 X SN74LV4053 is good

jcoleman-oneaihealth, 2023-11-12:
Why MUX every SpiRam chip pin ?
If the SpiRam is Not Enabled via its CS then the Chip IOs are disabled.

I dual connected each SpiRam to the same W806 GPIO execpt the CS pins were on separate GPIO pins (PB1 and PB6).
I could then Intitialize PSRAM and use SpiRam with CS->PB1 fine.

But when I took GPIO PB1 to Imput , then Initialized PSRAM with CS=GB6 , 
 -- the PSRAM just went to retuning FFs.

AnatolSher, 2023-11-12:
> Why MUX every SpiRam chip pin to the GPIOs ?
> If the SpiRam is Not Enabled via its CS - then all of that Chip IOs are disabled.

First: Because I'm not sure that the PSRAM pins of the chip (d0-d3) will go to Z state when the CS signal is high.
Second: You cannot influence the CS line of the PSRAM memory controller

Take a good logic analyzer and look in real time at what is happening at the pins of your dual assembly. Especially watch the CS line

jcoleman-oneaihealth, 2023-11-12:
Yeah, got that.
CS lines are tied High via 10K resistors.

I think it needs something to De-Initialize current PSRAM # 1 with CS=PB1,
then i can Initialize 2nd PSRAM(CS=PB6).

And vice-a-versa, to switch back to PSRAM # 1 :
De-Initialize current PSRAM # 2 (with CS=PB6),
then i can Initialize PSRAM # 1 (with CS=PB1) again.

It may be slower, but I can switch back and forth between PSRAMs.
Just need to keep Initializing next PSRAM # .

AnatolSher, 2023-11-12:
wm_psram.*
C:\Users\User\AppData\Local\Arduino15\packages\w80x_duino\hardware\XT804\0.0.5\cores\w806\include\driver
C:\Users\User\AppData\Local\Arduino15\packages\w80x_duino\hardware\XT804\0.0.5\cores\w806\lib\drivers

jcoleman-oneaihealth, 2023-11-12:
Ok Thx ,
let me try.
Maybe this would be helpful ? : 
```
  /// * @brief          This function is used disable psram.
  // __STATIC_INLINE void wm_tipc_disable_psram(void)
  wm_tipc_disable_psram();

```

AnatolSher, 2023-11-15:
**jcoleman-oneaihealth**
Hello! What progress has been made? You intrigued me with your idea with two psram chips. If I have free time next week, I'll try to do this in practice too. I don't know why I need this :)

jcoleman-oneaihealth, 2023-11-15:
I need it.
I am working on Robotics AI, and Vision Recognition.