board707/w80x_arduino #34: Board variants missing

ITstreet1, 2024-01-25:
Hi,
I have installed the latest version of the package support, but there are some boards missing. I have support for only 
W801, W805, W806, and Air103. I don't have for Air101, W800, and W802. I tried to reinstall the package, but it is the same.

Any suggestion?

Regards

board707, 2024-01-25:
Air101 should be there, w800/802 boards are not supported

ITstreet1, 2024-01-25:
Air101 is not there. I used this link
https://raw.githubusercontent.com/board707/w80x_arduino/hal-v0.6.0/package_w80x_index.json

ITstreet1, 2024-01-25:
But the JSON links say 0.6.0. ?

board707, 2024-01-25:
"0.6.0" is a version of HAL
The latest Arduino package has a version number 0.0.5

ITstreet1, 2024-01-25:
Anyway, made it manually. It works

ITstreet1, 2024-01-26:
Just noticed, no support for Air105. Any plans to add it?

board707, 2024-01-26:
As far I know Air105 based on completely different core and not compatible with WinnnerMicro SDK. So it's unlikely that it will ever be supported.

ITstreet1, 2024-01-26:
When should we expect the 0.6.0?

board707, 2024-01-26:
I can not tell. It depends not only on me, we have several people on the team. I think within 1-2 weeks

closed 2024-01-26 by ITstreet1

board707, 2024-01-29:
Version 0.0.6 released.