board707/w80x_arduino #38: I am a fan of Chinese.

Chenjiek, 2024-04-23:
Hello, I am a fan of Chinese. I used your w80x motherboard and Arduino programming to display a 0.96-inch TFT screen, but I tried many times and couldn't successfully display it completely. There is also a 1.3-inch screen with a 240 * 240 display and ILI9341 display, but the colors always don't match. Red will display yellow, blue will display light blue, and black will display white. I wonder if you can continue to update to allow w80x to support more screen displays. There is also a bug. When I use the default I2C, which is PA1/PA4, as I2C, and then use PA2 as AD input, it will get stuck and unable to initialize. I hope to... Waiting for your update to resolve all the above issues, thank you!

Chenjiek, 2024-04-23:
0.96 uses st7735s, 1.3 uses st7789

AnatolSher, 2024-04-23:
Hello!
A carelessly written sketch gives the following result:
[image]

#include "Arduino.h"
#include "Wire.h"
#include "LCD2004_PCF8547.h"
#include "stdlib_noniso.h"
#define LCD_WIDTH   16
#define LCD_HEIGHT  2

void setup() {

Wire.begin();    
LCDInit ();
delay(500);
LCDSetBacklightOn();
Serial.begin(115200);
pinMode(PA2,ANALOG_INPUT);

}

void loop() {
LCDClear ();
char string[25];
long volt = 0;
volt = analogRead(PA2);
Serial.printf("ADC millivolt [%f]\r\n", volt);
ltoa(volt,string,10);
LCDWriteDataString (0, 0, (char*)"    mV");
LCDWriteDataString (0, 0, string);
delay (1000);
}

AnatolSher, 2024-04-23:
This work with OLED via software i2c
[image]
 
[adc_oled_i2c.zip](https://github.com/board707/w80x_arduino/files/15082722/adc_oled_i2c.zip)

I don’t yet know why the libraries conflict. Maybe I'll investigate this glitch during this week.

Chenjiek, 2024-04-24:
I am glad to receive your reply. It is indeed possible to display and AD simultaneously using the one you sent. As I need to use the u8G2 library, the U8G2 library has a Chinese display that can be used directly. However, when using the U8G2 library, it will get stuck in "pinMode (PA2, ANALOGINPUT);" and the operation will stop. I don't know where the error is, please help me find a solution. Thank you!
//使用U8G2库硬件I2C
#include <U8g2lib.h>
//U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ PB6, /* data=*/ PB7, /* reset=*/ U8X8_PIN_NONE);//使用软件模拟I2C
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);//使用硬件I2C

void setup(void) {
  //u8g2.begin(/*Select=*/ A0, /*Right/Next=*/ 5, /*Left/Prev=*/ 9, /*Up=*/ 8, /*Down=*/ 10, /*Home/Cancel=*/ A1); // Arduboy DevKit
    Serial.begin(115200);
    Serial.println("start!");
  u8g2.begin();  

    pinMode(PA2,ANALOG_INPUT);//------------------------------------------------------
    Serial.println("go to loop!");
}

uint8_t m = 24;

void loop(void) {
  char m_str[3];
  strcpy(m_str, u8x8_u8toa(m, 2));		/* convert m to a string with two digits */
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_logisoso62_tn);
    u8g2.drawStr(0,63,"9");
    u8g2.drawStr(33,63,":");
    u8g2.drawStr(50,63,m_str);
  } while ( u8g2.nextPage() );
  delay(1000);
  m++;
  if ( m == 60 )
    m = 0;
    uint32_t volt=0;
volt = analogRead(PA2);
Serial.printf("%lu", volt); // 使用 %lu 格式化说明符打印无符号长整数值
Serial.println("");
}

Chenjiek, 2024-04-24:
This is a 1.3-inch screen of ST7789, 240 * 240, originally without CS pins. Due to the inability to display, I connected the CS of the screen. However, using your routine, the screen can display, but black will display white, red will display yellow, and so on. Many colors are not correct. As I am not very good, I do not know where to modify them? I will put the code up, can you help me take a look? Thank  @you!
`// ILI9341 library example
// (c) 2020 Pawel A. Hernik
//  PINOUT  X9C10X TOP VIEW  (see datasheet)
//  U/D   25
//  INC   24
//  CS    23
//  VCC   +5V
/* Caution! for w80x!
 ILI9341 240x320 3.2" V1.0 SPI LCD pinout (header at the top, from left):
 #1 MISO  -> NC
 #2 LED   -> 3.3V (5v)
 #3 SCK   -> PB6
 #4 SDI   -> PB7
 #5 DC    -> PB9
 #6 RESET -> PB10
 #7 CS    -> PB14
 #8 GND   -> GND
 #9 VCC   -> 3.3V (5v)
*/
#include "Adafruit_GFX.h"//屏幕显示的库
#include "ILI9341_Fast.h"//屏幕显示的库
#define SCR_WD 240//宽度
#define SCR_HT 240//高度
#define TFT_CS PB14
#define TFT_DC PB9
#define TFT_RST PB10

ILI9341 tft = ILI9341(TFT_DC, TFT_RST, TFT_CS);

void setup(void) {

  //u8g2.begin();
  pinMode(PA1, ANALOG_INPUT);  // or pinMode(PA2,ANALOG_INPUT);
  pinMode(PA2, ANALOG_INPUT);  // or pinMode(PA2,ANALOG_INPUT);
  Serial.begin(115200);
  Serial.println(F("ILI9341 240x240"));
  tft.begin();
  tft.fillScreen(BLACK);
  tft.setCursor(0, 0);
  tft.setTextColor(WHITE);
  tft.setTextSize(2);
  tft.println("ILI9341 240x320");
  tft.println("Library Benchmark");
  tft.println("starts in 3s ...");
  delay(3000);
  tft.fillScreen(BLACK);
  Serial.println(F("Benchmark   Time (microseconds)"));
}

void loop(void) {
  float volt = 0;
  float volt1 = 0;
  volt1 = analogRead(PA1);
  volt = analogRead(PA2);

  // 将浮点数转换为字符串
  char volt_str1[10];
  // 将浮点数转换为字符串
  char volt_str[10];
  sprintf(volt_str, "%.2f", volt);
  sprintf(volt_str1, "%.2f", volt1);

  // 比较旧的和新的字符串
  if (strcmp(volt_str, volt_strA) != 0) {
    tft.setCursor(10, 10);
    tft.setTextColor(BLACK);  // 设置文字和背景颜色相同
    tft.setTextSize(5);
    tft.println(volt_strA);  // 用背景色打印一遍旧内容以清除
    tft.setCursor(10, 10);
    tft.setTextColor(RED);
    tft.setTextSize(5);
    tft.println(volt_str);
    // 更新存储的旧内容字符串
  }
  strcpy(volt_strA, volt_str);  // 使用strcpy复制字符串

  if (strcmp(volt_str1, volt_strB) != 0) {
    tft.setCursor(10, 50);
    tft.setTextColor(BLACK);  // 设置文字和背景颜色相同
    tft.setTextSize(5);
    tft.println(volt_strB);  // 用背景色打印一遍旧内容以清除
    tft.setCursor(10, 50);
    tft.setTextColor(RED);
    tft.setTextSize(5);
    tft.println(volt_str1);
    strcpy(volt_strB, volt_str1);  // 使用strcpy复制字符串
    // 更新存储的旧内容字符串
  }
 

  delay(100);  // 稍作延迟，用于按钮防抖
}
`

AnatolSher, 2024-04-24:
Hello! To avoid wasting time, watch this discussion. There is a discussion about a fast library for w80x
http://ask.winnermicro.com/question/692.html

AnatolSher, 2024-04-24:
About U8G2 Lib ver. 2.34.22
Constructor
[image]
Code
[image]
Result
[image]
In software i2c emulation mode it work

AnatolSher, 2024-04-24:
Of course! 
Dont worry, be happy :)
Little bit later...

Chenjiek, 2024-04-24:
Really? Thank you so much, I'm looking forward to it......

AnatolSher, 2024-04-24:
So,screen color fill is correct
[Test_ili9341_sdio_spi.zip](https://github.com/board707/w80x_arduino/files/15097461/Test_ili9341_sdio_spi.zip)

Chenjiek, 2024-04-25:
[image]
[image]
[image]
[image]
The screen color applied to ILI9341 is correct. What I mean is whether it is possible to add a routine for st7789 with a size of 1.3-inch 240 * 240. As using this library directly on a 1.3-inch screen may result in color errors, and adding support for a 0.96-inch 160 * 80 tft screen would be better to add a program that can be used directly in the routine.

Chenjiek, 2024-04-25:
I tested using the Arduino_ST7789-Fast library and encountered many errors, which was too difficult for me.

Chenjiek, 2024-04-25:
I just used the above program and the screen displayed the correct color smoothly for the first time. It's so beautiful, thank you!

AnatolSher, 2024-04-26:
Hello! This works with the driver for ST7789. You need to play with the WidthXHeight parameters. And maybe - rotation
[image]

Chenjiek, 2024-04-26:
[image]
[image]
Hello, I used the ST7789 library above to drive a 0.96-inch TFT screen and found that the color is incorrect. Red will display blue, blue will display red, and green seems to be normal. Moreover, its position is not in the correct position. Where can I modify it to achieve the correct color and position?

Chenjiek, 2024-04-26:
[image]

Chenjiek, 2024-04-26:
[image]

AnatolSher, 2024-04-26:
Yes, you are right. The color is distorted. Okay... Now let's see what's going on with the initialization process

AnatolSher, 2024-04-26:
Try this
[ST7735_AdafruitBenchmark.zip](https://github.com/board707/w80x_arduino/files/15129348/ST7735_AdafruitBenchmark.zip)
Caution! You need to independently recalculate the coordinates and offsets in the setRotation method for different screen positions. I didn't have the patience :)

AnatolSher, 2024-04-26:
After testing, close this issue. Give us a star :) and open a new issue with the wording "The system crashes when ADC and hw i2c are used at the same time". This is a separate issue and there may be a bug in the HAL... We are working on it

Chenjiek, 2024-04-26:
I have seen your reply, thank you. As it is already evening here, I will do as you said tomorrow. I will publish these in the QQ group of WinnerMicro and recommend everyone to use these libraries together. Thank you!

closed 2024-04-27 by Chenjiek