board707/w80x_arduino #39: The system crashes when ADC and hw i2c are used at the same time

Chenjiek, 2024-04-27:
Thank you for helping me solve the problem. I am currently encountering a situation where the LED screen SSD1306 of U8G2 using PA1 and PA4 as hardware I2C freezes when reading the ADC voltage. You can use software to simulate I2C at other pins, but the refresh rate displayed is too slow. We look forward to resolving this issue soon. Thank you!

AnatolSher, 2024-04-28:
你好 :)
Replace these files in your Arduino package, test and tell us the result
[Change_this.zip](https://github.com/board707/w80x_arduino/files/15143592/Change_this.zip)

Chenjiek, 2024-04-29:
Okay, I did a simple compilation and encountered an error. I will send the error message below:
C:\Users\Administrator\AppData\Local\Arduino15\packages\w80x_duino\hardware\XT804\0.0.6\cores\w806\HardwareI2C.cpp:2:25: fatal error: HardwareI2C.h: No such file or directory
 #include "HardwareI2C.h"
                         ^
compilation terminated.

exit status 1

Compilation error: exit status 1

AnatolSher, 2024-04-29:
Hi! If you are using Windows, you need to rewrite the files in the following folders:
C:\Users\User\AppData\Local\Arduino15\packages\w80x_duino\hardware\XT804\0.0.6\libraries\Wire <- HardwareI2C.cpp
C:\Users\User\AppData\Local\Arduino15\packages\w80x_duino\hardware\XT804\0.0.6\cores\w806 <- Arduino.c
C:\Users\User\AppData\Local\Arduino15\packages\w80x_duino\hardware\XT804\0.0.6\cores\w806\lib\drivers <- adc.c
Where User - your login in Windows

Chenjiek, 2024-04-29:
I didn't understand what you meant, I don't know where to start, sorry, I can't understand......

AnatolSher, 2024-04-29:
For maximum speed hw i2c
[image]

Chenjiek, 2024-04-29:
[image]
[image]
Yes, I tested it for a long time just now and it seems that the conflict between AD input and I2C has been resolved. The screen refresh speed is very fast, which can be described as very, very beautiful. Thank you very much for your help. Thank you, but I have a question. Why is the lowest voltage of this ADC not 0V, but 19mv?

AnatolSher, 2024-04-29:
Hmm... Very beautiful girl :)

To use ADC correctly, you need to study Chapter 26 of the datasheet for w80x chips. Individual calibration is required for accurate measurements. The calibration technique was described in a question on ask.winnermicro.com. I don't remember exactly. HAL provides simplified ADC management. More advanced control is implemented in the W800-SDK provided by WinnerMicro.

Chenjiek, 2024-04-29:
Okay, it looks like some advanced knowledge. Thank you. This has already been completed. Do I also need to close it?

closed 2024-04-29 by Chenjiek