board707/w80x_arduino #40: Another concern is when can we achieve support for Arduino's WiFiManager library and Bluetooth related libraries?

Chenjiek, 2024-04-28:
Another concern is when can we achieve support for Arduino's WiFiManager library and Bluetooth related libraries?
Yesterday, I shared information and screen display of Arduino's W80X on GITHUB in several relevant QQ groups. Many friends asked me when WIFI is supported?

AnatolSher, 2024-04-28:
Hello! This is a good question and we are working on it currently. This project is called w80x-arduino-iot. But we are not ready to publish it yet. There are still many features to be implemented. Follow our github :)

Chenjiek, 2024-04-29:
Okay, I look forward to using w801 using WIFI. Thank you for your efforts!

divadiow, 2024-05-14:
I look forward to wifi support!

I read somewhere that W806 actually has the same core as W801 and wifi can be enable if using the W801 SDK. 

http://ask.winnermicro.com/article/186.html

is this true?

AnatolSher, 2024-05-14:
This discussion does not talk about a proven way to enable the RF module. The author gives a recipe on how to use WM80X-SDK on W806 chips. This is due to a change in the chip's memory map. Our compatriot nvv13 made a special SDK build for this purpose. In theory, all W80X chips are made from one lithographic mask. When monitoring parameters, restrictions are made on operating modes. We have not done any experiments on RF activation on W806 chips. Because it doesn't make sense.
https://github.com/nvv13/test/tree/main/test-mk/w806/W806_SDK-linux

AnatolSher, 2024-12-19:
@Chenjiek Hello
You can close this issue because the WiFi plugin is ready and can be downloaded from https://github.com/board707/w80x_arduino_iot.
You can learn how to use it here http://ask.winnermicro.com/article/212.html

Chenjiek, 2024-12-24:
Received, I will learn how to use it as soon as possible, thank you

Chenjiek, 2024-12-24:
[image]
[image]
Is my path correct? Why don't I seem to know what the effect is and how to use it?

divadiow, 2024-12-24:
So is WiFi on W806 possible or not?

divadiow, 2024-12-24:
I am trying to find out if it even does

AnatolSher, 2024-12-24:
@Chenjiek Path not correct. Put it to home directory "C:\Users\User\Documents\Arduino\hardware" like this
[image]
WiFi have only w800,w801,w802,w803

AnatolSher, 2024-12-24:
@divadiow We've discussed this topic before. There is probably an RF module in there, because the lithographic mask is probably one for all w80x chips. And even if it is, you won't make it work. Because there is no factory calibration data. And even if you find that data, you can't write it to RO memory. I could go on with that logic :) ...  You also have to take care of the antenna and its impedance

divadiow, 2024-12-24:
Sure. I have officially given up on W806 with WiFi :)

Chenjiek, 2024-12-25:
[image]
Thank you, my friend, I don’t seem to have found the same folder as you. There is no hardware folder in my folder. Do I need to create one myself?

AnatolSher, 2024-12-25:
@Chenjiek Yes

Chenjiek, 2024-12-25:
OK, I created the new folder and it's ready to use, thank you! Wish you a happy new year!

closed 2024-12-26 by Chenjiek